UCI-HyperXite/inverter-control #11: Deadtime incorporation in SPDM code

SM-dot, 2024-01-26:
Deadtime - adding a delay when going from low to high(time TBD) to avoid short circuiting. 

Basically, we can check if previously it was low and the current state is high then we can delay for X nanoseconds. In case of high to low, no time delay is needed as it is safe(as discussed with professor Smedley).

taesungh, 2024-02-06:
For reference, the two outputs for each phase should never both be high at the same time:
[image]

SM-dot, 2024-02-06:
Hi, for this - this is what I have so far:
if (prev_state == 0 && Vu == 1)
{
        sleep_us(5); //time TBD 
        prev_state += 1;
}

basically keeping track of the states as recommended by Chris. Do you have any other suggestions/ plans of implementation?

taesungh, 2024-03-14:
From discussion in lab, the gate driver has a rising time and propagation delay that is longer than the falling time and propagation delay, so we shouldn't need to worry about implementing dead time on the software side.

closed 2024-03-14 by taesungh